anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #1: Docker is not working on R4S

weikai, 2021-05-20:
Is docker working in the latest build?  I got the following error message from dockerd on R4S.

`Thu May 20 16:05:29 2021 daemon.err dockerd[7612]: main.(*DaemonCli).start(0x4000c8d170, 0x40007fc5a0, 0x0, 0x0)
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]:      /home/runner/work/OpenWrt-NanoPi-R2S-R4S-Builds/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/dockerd-20.10.6/.go_work/build/src/github.com/docker/docker/cmd/dockerd/daemon.go:195 +0x588
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]: main.runDaemon(...)
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]:      /home/runner/work/OpenWrt-NanoPi-R2S-R4S-Builds/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/dockerd-20.10.6/.go_work/build/src/github.com/docker/docker/cmd/dockerd/docker_unix.go:13
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]: main.newDaemonCommand.func1(0x4000cc0000, 0x40007fad30, 0x0, 0x1, 0x0, 0x0)
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]:      /home/runner/work/OpenWrt-NanoPi-R2S-R4S-Builds/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/dockerd-20.10.6/.go_work/build/src/github.com/docker/docker/cmd/dockerd/docker.go:34 +0x78
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]: github.com/docker/docker/vendor/github.com/spf13/cobra.(*Command).execute(0x4000cc0000, 0x400004e050, 0x1, 0x1, 0x4000cc0000, 0x400004e050)
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]:      /home/runner/work/OpenWrt-NanoPi-R2S-R4S-Builds/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/dockerd-20.10.6/.go_work/build/src/github.com/docker/docker/vendor/github.com/spf13/cobra/command.go:850 +0x320
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]: github.com/docker/docker/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0x4000cc0000, 0x0, 0x0, 0x10)
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]:      /home/runner/work/OpenWrt-NanoPi-R2S-R4S-Builds/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/dockerd-20.10.6/.go_work/build/src/github.com/docker/docker/vendor/github.com/spf13/cobra/command.go:958 +0x258
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]: github.com/docker/docker/vendor/github.com/spf13/cobra.(*Command).Execute(...)
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]:      /home/runner/work/OpenWrt-NanoPi-R2S-R4S-Builds/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/dockerd-20.10.6/.go_work/build/src/github.com/docker/docker/vendor/github.com/spf13/cobra/command.go:895
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]: main.main()
Thu May 20 16:05:29 2021 daemon.err dockerd[7612]:      /home/runner/work/OpenWrt-NanoPi-R2S-R4S-Builds/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/dockerd-20.10.6/.go_work/build/src/github.com/docker/docker/cmd/dockerd/docker.go:97 +0x188`

anaelorlinski, 2021-05-21:
I didn't test it, just added the Openwrt packages, so there might be missing kernel options or missing packages, but those logs don't give a clue.
I will try to find what's wrong when I get a bit of spare time.

anaelorlinski, 2021-05-21:
Added kernel config options found here :
https://gitlab.com/mcbridematt/openwrt-container-feed/-/blob/master/openwrt_configs/openwrt-docker-host-config-x86_64

Hope it helps

anaelorlinski, 2021-05-22:
`root@OpenWrt:/# docker run -it --rm -d -p 8080:80 nginx
Unable to find image 'nginx:latest' locally
latest: Pulling from library/nginx
fcad0c936ea5: Pull complete 
d5c473a8294d: Pull complete 
8e31422295d0: Pull complete 
209e8d4861a8: Pull complete 
3314b286b1b3: Pull complete 
574020532afe: Pull complete 
Digest: sha256:df13abe416e37eb3db4722840dd479b00ba193ac6606e7902331dcea50f4f1f2
Status: Downloaded newer image for nginx:latest
6a01848f7dd779d0167385789c236dcae3dda8a9228490e7c7ff0fc4dccc233b

root@OpenWrt:/# docker ps
CONTAINER ID   IMAGE     COMMAND                  CREATED          STATUS          PORTS                                   NAMES7 
  nginx     "/docker-entrypoin.…"   13 seconds ago   Up 10 snds   0.0.0.0:8080->80/tcp, :::8080->80/tcp   stoic_kirch

root@OpenWrt:/# curl localhost:8080
<!DOCTYPE html>
<html>
<head>
<title>Welcome to nginx!</title>
<style>
    body {
        width: 35em;
        margin: 0 ont-family: Tahoma, Verdana, Arial, sans-serif;
    }
</style>
</head>
<body>
<h1>Welcome to nginx!</h1>
<p>If you see this page, the nginx web server is successfully installed and
working. Further configuration is required.</p>

<p>For online documentation and support please refer to
<a href="http://nginx.org/">nginx.org</a>.<br/>
Commercial support is available at
<a href="http://nginx.com/">nginx.com</a>.</p>

<p><em>Thank you for using nginx.</em></p>
</body>
</html>
root@OpenWrt:/#`

anaelorlinski, 2021-05-22:
tested working on r2s

closed 2021-05-22 by anaelorlinski

weikai, 2021-05-23:
@anaelorlinski - It works great.  Thank you!