anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #2: Booting Issues with Full Squashfs Build

weikai, 2021-05-23:
It looked like there're booting issue on the latest full squshfs build.  I was able to run sysupgrade using NanoPi-r4s-squashfs-full.zip and made docker working.  I decided to run sysupgrade with -n option on the same version without keeping the configuration.  It looked like the old configuration never got deleted so I decided to reflash the sd card with the full version.  It failed to boot this time.  I tried the mini version, it booted.  However, when I run sysupgrade again in mini version to upgrade to the full version, it failed to boot again.  

I'm currently compiling a local version with the code checked out from Github to see if the problem is on Actions or it's somewhere else.

weikai, 2021-05-23:
ok,  changed kernel part size to 40M fixed the issue.

diff --git a/seed/ao.seed b/seed/ao.seed
index ccd00ed..f50a00c 100644
--- a/seed/ao.seed
+++ b/seed/ao.seed
@@ -449,6 +449,6 @@ CONFIG_PACKAGE_zlib=y
 CONFIG_PACKAGE_zoneinfo-core=y
 CONFIG_PACKAGE_zoneinfo-europe=y
 CONFIG_SOCAT_SSL=y
-CONFIG_TARGET_KERNEL_PARTSIZE=20
+CONFIG_TARGET_KERNEL_PARTSIZE=40
 CONFIG_TARGET_OPTIONS=y
 CONFIG_TARGET_ROOTFS_PARTSIZE=1024

anaelorlinski, 2021-05-23:
From my previous experiments while initially testing my builds images I also noticed that reflashing with squashs does not necessarily reset the settings, because the settings partition is not erased.
I am not sure that your change the kernel partition size will fix it. I think it will erase the settings partition it on next flash, but later the problem will happen again.
A good cleanup option (if the system is still booting) is to run firstboot and reboot to reset the overlay partition

weikai, 2021-05-24:
@anaelorlinski - The full version did not have enough space for kernel and failed to boot.  Increased the kernel part size fixed the boot issue.

anaelorlinski, 2021-05-25:
Oh got it. I could not imagine the kernel would be that big.

anaelorlinski, 2021-05-25:
Patch added. Thanks for contribution

closed 2021-05-25 by anaelorlinski