anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #9: WAN ethernet adapter does not show up

olesalscheider, 2021-12-14:
I just flashed the latest build on my R4S, but the WAN ethernet adapter (rtl8211e) does not show up in "ifconfig -a".
Is it supposed to work with this build, or does it need another kernel module?

anaelorlinski, 2021-12-14:
I did not test the latest release. However the release OpenWrt 21.02 2021.10.11 has proven to work perfectly so you might want to flash it.
I will give a try with latest release in the next days.

olesalscheider, 2021-12-14:
I just tried the older build, but I have the same problem. The official build from the OpenWRT project page works though.

anaelorlinski, 2021-12-15:
Running OpenWrt 21.02 2021.12.14 as a router without problem on 4GB NanoPi r4s.
In my build the network driver is the one from realtek instead of default linux one, so you might check your configuration.

closed 2021-12-15 by anaelorlinski