anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #12: Mini Configured with LuCI?

gary-hammer, 2021-12-16:
Not actually an issue, but a question.
Is LuCI supposed to be functional after booting a mini build?
I installed OpenWrt-AO-21.02-20211214.2321-NanoPi-r4s-ext4-mini and after all LEDs were on I see that I do pull an IP via DHCP, but the web interface is just a directory listing showing a single folder that can't be entered.

Thanks for your work, this is a great system for OpenWRT

dapperdrop, 2021-12-16:
Mini version is command-line only. You'll want the full version for LuCI.

gary-hammer, 2021-12-16:
If I catch it at the right time, I may try a CLI install of LuCI.
I just didn't want the docker functionality in the build, so, chose mini.
So, both you and looking at the seed answered my question.

Thanks

closed 2021-12-16 by gary-hammer