anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #14: StrongSwan/LibreSwan support

aw1cks, 2021-12-23:
Hi there! Cool project, thanks! I made something similar but yours is better :)

I was wondering, if you would be opposed to adding a few kernel modules, needed for IPsec.
The following kernel modules would need adding:

- kmod-ip6-vti
- kmod-ipsec6
- kmod-ipt-ipsec
- kmod-ip-vti
- kmod-iptunnel4
- kmod-ipsec4
- kmod-xfrm-interface

I'll happily make a PR to do this - just wanting to check if it will be accepted if I do so!

anaelorlinski, 2021-12-23:
Sounds legit to me to be added in the full release, but only adding kernel modules will probably not be enough. Instead I could add those packages :
- libreswan : will automatically add the required kernel deps
- strongswan : seems heavy but modular so I will pick a subset of all available modules

aw1cks, 2021-12-23:
Sounds good! I'll make a PR.

anaelorlinski, 2021-12-23:
I added libreswan (and dependencies). Build is in progress.
Regarding strongswan, it cannot be built-in at the same time as libreswan so I made a choice because there are much less dependencies.

aw1cks, 2021-12-24:
Thanks - package is showing up, I will test it out over the next few days. Happy holidays!

closed 2022-01-02 by anaelorlinski