anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #16: Issues resizing root partition

socketfork, 2022-01-05:
Hello there,

I've been taking a look at your builds for the NanoPi R4S.  I seem to be having issues resizing the root partition that are specific to the image **OpenWrt-AO-21.02-20211224.0120-NanoPi-r2s-squashfs-mini.zip**

I've attempted resizing locally using the fdisk and resize2fs method.  It's unable to perform the online resize.  So I also tried resizing the partition using the SD card in another machine using Gparted and it gives me a similar error as well.  I've tried running fsck on it as well, which gave me errors related to the superblock.  

I've used these two methods successfully with other builds, but for some reason it doesn't seem to be working here.  I was wondering if you have noticed this issue.

socketfork, 2022-01-05:
This issue doesn't seem to happen with the "full" build, only the "mini".  

Also here is the error log from Gparted:
```

Grow /dev/sdb2 from 1.00 GiB to 29.30 GiB  00:00:01    ( ERROR )
    
calibrate /dev/sdb2  00:00:01    ( SUCCESS )
    
path: /dev/sdb2 (partition)
start: 131072
end: 2228223
size: 2097152 (1.00 GiB)
grow partition from 1.00 GiB to 29.30 GiB  00:00:00    ( SUCCESS )
    
old start: 131072
old end: 2228223
old size: 2097152 (1.00 GiB)
new start: 131072
new end: 61571071
new size: 61440000 (29.30 GiB)
grow file system to fill the partition  00:00:00    ( ERROR )
    
resize2fs -p '/dev/sdb2'  00:00:00    ( ERROR )
    
Filesystem at /dev/sdb2 is mounted on /media/parallels/rootfs; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 2
resize2fs 1.45.5 (07-Jan-2020)
resize2fs: Invalid argument While checking for on-line resizing support

```

dominch, 2022-02-04:
This issue still needs to be resolved, commands from openwrt docs don't work (partition in use), eventually I was able to resize partition but it was few steps and this have to be repeated on each flash. It's bit easier to do that on sd card reader connected to different linux. 
I think maybe start partition should be bigger? Right now it's hard to find micro SD smaller than 8GB. Increasing CONFIG_TARGET_KERNEL_PARTSIZE and CONFIG_TARGET_ROOTFS_PARTSIZE would help a lot, right now it's 1GB only 
Some systems use scripts to resize flashed image on first boot (RPI, Armbian), maybe it's possible here? I know that each router model has fixed resources (and they are usually small) but here we can any card.

anaelorlinski, 2022-02-05:
I will give a try increasing the default partition size to 8Gb. Hope the flash time will not be x8 ....

dominch, 2022-02-07:
It should not increase image or flashing time, the only thing that changes is partition size on first bytes of image :) Thanks. 
Are You resizing you partition after burning with some scripts?

anaelorlinski, 2022-02-11:
Increasing the ROOTFS size increases also flashing time by the same factor and this is not very satisfying.
I tried a solution that seems to work and archived it here : https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/blob/main/docs/resize-f2fs.md
Current images miss the cfdisk and losetup tool, this will be fixed soon.

anaelorlinski, 2022-02-12:
New images with missing tools are generated.
Follow the steps from there for squashfs images : https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/blob/main/docs/resize-f2fs.md

closed 2022-02-12 by anaelorlinski

dominch, 2022-02-14:
Thanks, I'll try it soon, 
Are those for SquashFs only? Have You tried with ext4? As far as I remember I got "filesystem busy" and needed to unmount fs first. Would they work on both mini and full version?

anaelorlinski, 2022-02-14:
I did not try ext4 resizing, in the issue squashfs was mentionned.
Instructions can be found here I will give a try later https://openwrt.org/docs/guide-user/installation/openwrt_x86#resizing_filesystem
However resizing offline ext4 can be done easily on any linux system, so this might be easier.

dominch, 2022-02-16:
Squashfs was designed for routers with very limited resources, here ext4 should be better option, it's more like small computer with great resources and there is no need to decompress it on fly and do overlays. I know I can resize partition on linux and this is easiest to do, on windows much more work is needed (mainly to get working sd card reader). 
I think about reasonable system partition (that's why I asked for bit more) and data partition (mainly for docker). That would help with upgrades, 
thanks for Your work :)