anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #17: Unable to install some packages

pcartwright81, 2022-02-14:
cannot find dependency kernel (= 5.4.179-1-a76fa007c9d58d46d6f01d54cf13ae72)
uname -r is 5.4.171
opkg update is pointing to https://downloads.openwrt.org/releases/21.02-SNAPSHOT for all.
Installed via release OpenWrt-AO-NanoPiR4S-full-21.02-20220212-ext4

anaelorlinski, 2022-02-14:
Adding extra packages with okpg is not supposed to work our of the box with those builds (same problem with official openwrt snapshots).
However sometimes by luck it will work anyway as long as the package does not depend on kernel modules (because kernel hash is different from snapshot builds). You can also try to force install if the kernel module you require is already in the build.
The proper solution to add packages to this build can be :
- fork this repository, update .seed files according to your wishes and generate your own build
- let me know the package you miss, I might include it in the build if I think it could be useful to others

closed 2022-02-14 by anaelorlinski

pcartwright81, 2022-02-14:
Thanks for the reminder.   I recently got back into custom OpenWrt  with the R4S.   Most of the packages relate to USB storage and the specific package is kmod-fuse.  I have a drive that is NTFS which I do not want to switch right now.