anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #21: SQM upload and download mixed

twiikker, 2022-03-28:
Latest full version for Nanopi R4S.
So i just installed it yesterday and noticed that upload and download speed settings are mixed.
If i put them correctly for my 1000/500 connection (900000 for dl and 490000 for up) i get 470/470 and if i set them other way around i get 870/470.

anaelorlinski, 2022-03-28:
Thanks for reporting, however I am not sure this issue is related to my build or a more general OpenWrt issue. I would recommend to check on OpenWrt forums or open an issue on OpenWrt github repository. Let me know if there is a fix/workaround and I could integrate it to the build.

shark873, 2022-03-28:
> Latest full version for Nanopi R4S. So i just installed it yesterday and noticed that upload and download speed settings are mixed. If i put them correctly for my 1000/500 connection (900000 for dl and 490000 for up) i get 470/470 and if i set them other way around i get 870/470.

This is happening cause by default the interface that comes selected in SQM page is eth1 and not eth0.

closed 2022-04-04 by anaelorlinski