anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #22: Build failing, can you check if yours fails too?

shark873, 2022-03-28:
I cloned your repo, i was using it to run some builds... but the past few days they are failing when building a determinate package can you check if yours will fail too?

Amazing work here btw, thx.

anaelorlinski, 2022-03-28:
Started a new build. Let's see what happens.

anaelorlinski, 2022-04-04:
My builds did complete. Probably your .config seed is buggy or you are building extra packages that are broken.

closed 2022-04-04 by anaelorlinski

shark873, 2022-04-09:
Before the issue was related to -> https://github.com/openwrt/packages/issues/18177 <- which got fixed...

Now i have a new issue:

https://i.imgur.com/zHd1PUd.png

I have just re-done my .seed file but its still happening. Do you have any ideas? Thx.