anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #23: Is it possible to use your build on the R2C

snippem, 2022-04-04:
I ordered a R2C and when compared to the R2S the difference is in the network chip.
I read something about a patch to get the YT8521S working.
Would it be possible to integrate the patch in the R2S build?

anaelorlinski, 2022-04-04:
R2C will probably not work out of the box with R2S build, or miss network driver as you mention. However ImmortalWrt has support for it and this is from where I got most of the useful patches (uboot + linux kernel + drivers).
I will work on it, just be patient and I will update my repo to be able to build a R2C image. However I don't own a R2C so I will not be able to test it.

snippem, 2022-04-04:
> R2C will probably not work out of the box with R2S build, or miss network driver as you mention. However ImmortalWrt has support for it and this is from where I got most of the useful patches (uboot [...]

The fact you consider to do this is amazing:-)
I just ordered the device so i don't own the device yet. (Hope there will be more owners that are willing to test your builds. As I am not an internet expert haha)

anaelorlinski, 2022-04-12:
I have generated some builds for R2C. Hopefully they work, let me know if you get some success.

snippem, 2022-04-13:
> I have generated some builds for R2C. Hopefully they work, let me know if you get some success.

My device is shipped as soon as I receive the r2c I will let you know 👍

snippem, 2022-04-21:
Received the r2c .
Flashed full image with balena etcher attached only the lan side and powered up.
The device boots but askes for a password normally you don't have to fill anything in.
Only this time it wants a password so i stranded on the welkome screen for logging in.

anaelorlinski, 2022-04-21:
This sounds strange to be asked for a password. ssh root@192.168.1.1 should not. Which image did you flash ? mini/full ? squashfs/ext4 ?
There might be something missing still but at least it seems that at least the bootloader is fine.

snippem, 2022-04-21:
Flashed the full build ext4 version. Did a brief check of it booted then I had to go to work.
This was the first thing I ran into . I reflashed the card and when booted root wasn't sufficient to login. It states that I had the wrong password.

snippem, 2022-04-21:
Is it possible to use the backup from my R4S for the R2C ?
Hope it is possible it makes switching between devices easier to test if the firmware works for the R2C 😄

anaelorlinski, 2022-04-21:
For my own purpose I wrote a script based on uci commands over ssh and scp commands that recreates my router configuration from a clean image after boot. Using this I can switch seamlessly from R2S to R4S. So I think you can take most of the /etc/config files and others from /etc and this will work.

snippem, 2022-04-21:
Tried to reset with the resetbutton holding it for 10 minutes.
But this did not work either keeps saying Invalid username and/or password! Please try again.

anaelorlinski, 2022-04-21:
Just reflash the sdcard .... much faster. And try with squashfs

snippem, 2022-04-21:
> Just reflash the sdcard .... much faster. And try with squashfs

Flashed with squashfs same result password incorrect.
Wired to a different laptop same result incorrect username or password

snippem, 2022-04-21:
> Just reflash the sdcard .... much faster. And try with squashfs

The firmware i flashed is this one OpenWrt-AO-NanoPiR2C-full-21.02-20220412-squashfs.img.gz
Found out that the username is root and the password =password.
And the firmware image is friendlywrt when it is loaded haha was this intended?

anaelorlinski, 2022-04-21:
There must be something wrong then, my build has nothing to do with FriendlyWrt. I could be ImmortalWrt if there was a mistake in the patches but FriendlyWrt is simply impossible. Flashing must have been gone wrong at some point. Did you previously flash FriendlyWrt on the card ?
Also can you try to boot without SD card just in case ? Might have a clue if it still boots without a card inserted.

snippem, 2022-04-21:
I received the plus version with 8 GB off emmc haha I orderd the non plus🙃🙃🙃🙃😄😄😭😭

anaelorlinski, 2022-04-21:
that was my guess. might need to check to boot from TF card always https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R2C_Plus#The_Boot_order_between_eMMC_and_SD_card or flash to eMMC

snippem, 2022-04-21:
> that was my guess. might need to check to boot from TF card always https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R2C_Plus#The_Boot_order_between_eMMC_and_SD_card or flash to eMMC

Is it possible to flash directly on the emmc?

snippem, 2022-04-22:
Okay I messed up now the r2c does not boot anymore.
Any idea where the bootloader pins are located 😄

snippem, 2022-04-22:
Okay got booting from the sdcard working.
But not very handy with shorting the mask pinns.
So i didnt destoyed the board but now i have short the mask pinns not very handy .
I booted to the image you compiled and that worked fine.
But i need to sort out how to boot from the emmc or permanently from sd.
Do you have any experience in how to do that?

anaelorlinski, 2022-04-22:
from the doc it says that you should erase eMMC so it default boots to SD card with

dd if=/dev/zero of=/dev/mmcblk0 bs=8M count=1

just check if mmcblk0 is really the eMMC and not the SD card, because on my systems without eMMC the SD card is /dev/mmcblk0 so it might be /dev/mmcblk1 too

anaelorlinski, 2022-04-22:
> > that was my guess. might need to check to boot from TF card always https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R2C_Plus#The_Boot_order_between_eMMC_and_SD_card or flash to eMMC
> 
> Is it possible to flash directly on the emmc?

Probably but I have no experience here! I would say you need to boot on a SD card to have a system ready, download the image on the device, and flash it with dd

snippem, 2022-04-22:
Send a email to the support department off friendly. I have ask a proper instruction to or set a boot order with SD card first. Because shorting the pinns work but I have to remover the PCB out off its housing. Consider this the not so handy way. The other method does not work.
And ask how to flash your rom to the without bricking the device. It has something to do with the bootloader that is used from rockpi it has to be changed to a other one.
Will report back👍

snippem, 2022-04-23:
@anaelorlinski have setup the r2c with friendlywrt and the IPv6 is working now. Did this to make sure IPv6 is working so it could be a setting or a bug.

closed 2022-04-28 by anaelorlinski