anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #24: Downloads failing (R4S full build)

snippem, 2022-04-16:
If I had hair I would pull it out 😂
The problem that I have is that when I download firmware when let's say.
https://www.asus.com/nl/Networking-IoT-Servers/WiFi-Routers/ASUS-Gaming-Routers/RT-AX86U/HelpDesk_BIOS/
The download starts but then I get a network error. This is on both wired and wireless. I thought that it was site related but I also get this error on
the Asus Merlin download site https://sourceforge.net/projects/asuswrt-merlin/files/
I flashed your build that way it was clean  and without changes I made but the issue still persists.
Do you have any idea what could cause this issue?
Hardware wise I have 2 switches from TP-link managed (wireless excluded)

snippem, 2022-04-16:
Did some tinkering and when I remove the IPv6 wan everything works.
Downloading on the mentioned sites works as expected and much faster .

anaelorlinski, 2022-04-22:
I have read that software offloading fails with ipv6 on 21.02. Might be related ?

snippem, 2022-04-22:
Okay I have that enabled going to turnoff the setting.
Will report back👍

snippem, 2022-04-22:
> I have read that software flow offloading fails with ipv6 on 21.02. Might be related ?

Turned off software offloading this did not solve the issue.
The same problem occurs with ipv6 enabled the router traffic stops temporarily when i download a file  and then resumes.
This generates network wide error . stopping ipv6 en disabling at boot solves this.
In the mean time i will setup the r2c with friendlywrt to see if it has the same issues.(as i am waiting for a reaction off friendlyelec on the other issue)
Taken a quick look ipv6 works i have a ipv6 adres handed out to the WAN6.
What i do see is that somethings are different i do not see a dual-stack lite interface.
Also software offloading is completely disabled in the firewall settings the option is removed in the interface. 
And the files are downloaded till now without the issues described as above.
Going to set it up the R2C the same as the R4C to see if the same issue occurs. (later on

snippem, 2022-04-23:
@anaelorlinski have setup the r2c with friendlywrt and the IPv6 is working now. Did this to make sure IPv6 is working so it could be a setting or a bug.

closed 2022-04-28 by anaelorlinski