anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #25: pbr link invalid

BlackMage2, 2022-04-16:
the pbr link  https://github.com/stangri/source.openwrt.melmac.netkage is invalid

BlackMage2, 2022-04-16:
and the actual build for R2S has full pppoe support?

anaelorlinski, 2022-04-18:
> and the actual build for R2S has full pppoe support?

in the full build yes

BlackMage2, 2022-04-18:
> > and the actual build for R2S has full pppoe support?
> 
> in the full build yes

thanks for the information :D

BlackMage2, 2022-04-19:
and differs r2s from r4s? 1gb ram is ok?

BlackMage2, 2022-04-19:
it works with RK3328 but i dont know with network adapter, most likely with the RTL8211 network chip with lower temperatures

anaelorlinski, 2022-04-19:
> > > and the actual build for R2S has full pppoe support?
> > 
> > 
> > in the full build yes
> 
> thanks for the information :D i receive my NanoPi R2S Mini Router in the Beginning of May 2022, i hope it works then
> 
> what gcc is used in branch 22.03 ?
> 
> i hope my glass fiber box works with it(glass fiber -> ethernet converter, but with own protocol, i had problems on other routers with it, only a series of fritzboxes known works with it)
> 
> and the ext4 image should works fine?

In branch 22.03 I think it is GCC10. But that branch is still experimental and I might change the configuration later.

anaelorlinski, 2022-04-19:
> and differs r2s from r4s? 1gb ram is ok?

Those are not the same models and CPU. 1gb ram is enough for router purpose with few extra features (VPN, ...).
More memory is recommended if you want to use docker for example.

Don't hesitate to ask all questions on forum.openwrt.org and please keep only specific questions related to my builds here.

BlackMage2, 2022-04-19:
what can i make with docker?

BlackMage2, 2022-04-19:
but i hope the pppoe connection works

closed 2022-04-28 by anaelorlinski