anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #30: Can't install pppoe

mikegleasonjr, 2022-06-17:
See screenshot...

[image]

Using latest build on a Nanopi R4S 1gb version.

Thank btw for this build, since the OpenWRT snapshot doesn't support the 1gb version.

shark873, 2022-06-18:
> See screenshot...
> 
> [image]
> 
> Using latest build on a Nanopi R4S 1gb version.
> 
> Thank btw for this build, since the OpenWRT snapshot doesn't support the 1gb version.

https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/issues/17#issuecomment-1038908409

mikegleasonjr, 2022-06-18:
Yeah in the meantime I learned how to compile your repo, I thank you very much, I have a working pppoe connection new 🙏👌👍

closed 2022-06-18 by mikegleasonjr