anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #36: Latest 22.03 Doesn't boot on the R4S 1GB model

mikegleasonjr, 2022-09-10:
(as the title suggest)

anaelorlinski, 2022-09-12:
Suspecting the UBoot update but unfortunately I have no 1Gb device to test

Let me know if this one is working : https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/releases/tag/OpenWrtAO-22.03-20220819

mikegleasonjr, 2022-09-12:
I'll try.

Is there a way I can compile an up to date version with the previous UBoot if it works?

I shouldn't have cheaped out in buying a 1GB model. At the same time,, never ever will I need more than 1GB of memory for my home router...

anaelorlinski, 2022-09-12:
If it's linked to uboot it's easy. Just replace openwrt-22.03/patches/package/uboot-rockchip with the one from branch 21.02

mikegleasonjr, 2022-09-12:
No dice either for the previous version.

I'll compile my own with UBoot from 21.02 and keep you informed!

Thanks

anaelorlinski, 2022-09-12:
That's not good news then because the version I pointed you to is supposed to be using the previous uboot.
Let me know if it works with that one : https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/releases/tag/OpenWrtAO-22.03-20220428
It was using the big patchset of immortalwrt, and later I simplified it to get closer to openwrt mainline.

mikegleasonjr, 2022-09-12:
So the image I compiled while replacing UBoot from 21.02 worked (it booted).

Why I want  to upgrade to 22.03 is that I want top redo my config fresh.

It is said that 22.03 upgraded to firewall4 (whatever that means). If I flash 22.03 now and re-do manually my config, will I be able to upgrade future 22.03 releases only by flashing while "keeping current config" or this release still is using things from 21.02?

First time I am upgrading a major version of OpenWRT, just asking..

anaelorlinski, 2022-09-12:
OK definitely linked to uboot upgrade. I will check if I can find what could be the problem, but not easy without the device.

If you configure the firewall using /etc/config/firewall or the UI there should not be major changes from 21.02 to 22.03. If you have custom firewall rules using iptables then it might be more complicated.

I cannot confirm that upgrades keeping config using my builds work properly. My strategy regarding upgrades is different, I maintain a shell script that connects to the R4S over ssh and recreates the config by running many uci commands and uploading files to /etc/

anaelorlinski, 2022-09-15:
By chance do you have a serial console attached ? I would like to check logs of the bootloader

mikegleasonjr, 2022-09-15:
Hey I will retry tonight with your builds. I think one of my SD cards was acting up.

In the meantime I was able to build my own and smoothly upgrade from 21 to 22 without any issues.

Sorry for the confusions and I'll keep you informed

anaelorlinski, 2022-09-21:
Could you try one of the latest builds to know if it boots ?

mikegleasonjr, 2022-09-21:
Hey I'll try tonight 🙂

rpp293, 2022-09-25:
@anaelorlinski  Hi, I am new to github and coding. However, I can download the build and flash to my SD card for testing. I have the R4S 1GB version.

rpp293, 2022-09-25:
@mikegleasonjr  Would you mind sharing your build that you used for your R4S 1GB ?

rpp293, 2022-09-25:
@anaelorlinski I just tried your build OpenWrt 22.03 2022.08.19 on my R4s 1GB and only the pwr light is on. The sys, lan and wan show no activity.

mikegleasonjr, 2022-09-25:
[openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz](https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/files/9640796/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz)

@rpp293 there you go. A mini version with Cake QOS, AdGuard, DDNS and Luci...

ManuVice, 2022-09-25:
There was a new commit from friendlywrt. I will check it in a few days if it works on my R4S 1GB
https://github.com/friendlyarm/friendlywrt/commit/55a44722b9fc6736cf3046f9b5fac16cfdc42383

anaelorlinski, 2022-09-27:
> @anaelorlinski
> 
>     * OpenWrt-AO-NanoPiR4S-mini-22.03-20220922-squashfs: doesn't boot, SYS light never comes up
> 
>     * OpenWrt-AO-NanoPiR4S-mini-22.03-20220919-squashfs: doesn't boot, SYS light never comes up
> 
>     * OpenWrt-AO-NanoPiR4S-mini-22.03-20220908-squashfs: doesn't boot, SYS light never comes up
> 
>     * OpenWrt-AO-NanoPiR4S-mini-22.03-20220830-squashfs: doesn't boot, SYS light never comes up
> 
>     * OpenWrt-AO-NanoPiR4S-mini-22.03-20220819-squashfs: boots !

Confirms that the uboot upgrade did broke 1Gb support. I will revert it for now since the added value of the new version is close to nothing, and I will try to grab a cheap 1Gb device to do further testing

mikegleasonjr, 2022-09-27:
@anaelorlinski 

Send me a paypal link to donate for the purchase

anaelorlinski, 2022-09-27:
> @anaelorlinski
> 
> Send me a paypal link to donate for the purchase

https://www.paypal.com/donate/?business=8XQTGXAHEAKPY&no_recurring=0&currency_code=EUR

anaelorlinski, 2022-09-27:
Thanks @mikegleasonjr for donation I ordered a 1Gb model

anaelorlinski, 2022-09-28:
OpenWrt 22.03 2022.09.27 should work properly in 1Gb model

rpp293, 2022-09-29:
@anaelorlinski  I plan to test on my spare 64GB SD card later this weekend. 

So these are the steps I use to flash the [OpenWrt-AO-NanoPiR4S-full-22.03-20220922-ext4.zip](https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/releases/download/OpenWrtAO-22.03-20220922/OpenWrt-AO-NanoPiR4S-full-22.03-20220922-ext4.zip) file

Use command prompt to reformat the disk partition (SD Card)
Diskpart
Select Disk 2 (example- select the disk number for the sd card)
Clean
Create partition primary
format fs=ntfs quick or 
Reformat as exFat - using gui - right click

https://www.balena.io/etcher/
Extract image file and use the above software in link to flash 

Are you using a different method to flash?

anaelorlinski, 2022-09-29:
> @anaelorlinski I plan to test on my spare 64GB SD card later this weekend.
>
> So these are the steps I use to flash the [...]

Please use the latest version available! i released 2022.09.28 since then
You don't need to format the card before, just use balena etcher to flash the card and everything should be fine!

ManuVice, 2022-10-08:
> * [ ]  check this : [friendlyarm/friendlywrt@55a4472](https://github.com/friendlyarm/friendlywrt/commit/55a44722b9fc6736cf3046f9b5fac16cfdc42383)

commit works fine with 1GB and 4GB version. tested both 😉

Silvanskii, 2022-10-12:
> Please use the latest version available! i released 2022.09.28 since then You don't need to format the card before, just use balena etcher to flash the card and everything should be fine!
Hello, I wrote to you by mail, I need help for a reward, please check the mail

anaelorlinski, 2022-10-19:
Latest release 2022.10.19 has updated u-boot to 2022.07 with latest friendlywrt patches
Tested it boots successfully on a R4S 1Gb device. Thanks @mikegleasonjr and @ManuVice

commit 9ba0740d (2022-09-27): revert uboot to fix #36
closed 2022-10-19 by anaelorlinski