anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #38: Slow pppoe speed on 22.03 from 28.09

mandgeo, 2022-10-15:
I'm using R4S 4GB version. Today I've tried the ext4 release from 28th of september. The transfer speed over 5Ghz wifi is about 90 Mbps up/down. On the older 21.02 I get around 300 Mbps. Tests made on the same server, with speedtest.net.

anaelorlinski, 2022-10-15:
I tried to update to 2022.09.28 on my side and it seems I have perf issues too with that build. I don't know the reason yet I need to investigate further.
However my R4S is running 2022.08.31 for a while and no perf problems. Can you try that one ? If you confirm it works I will check what changed in openwrt between the 2 releases

anaelorlinski, 2022-10-15:
I would suggest trying with a cable first. Wifi speed can be affected by many factors

mandgeo, 2022-10-15:
300 Mbps with build from 30.08. I'll try again with build from 28.09 on cable, but i don't think much can change in 5 minutes

mandgeo, 2022-10-15:
You where right. WIFI miracles. It's working fine now.

Sorry for bothering you for nothing.

closed 2022-10-16 by anaelorlinski