anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #39: r8152 crashing on 21.02 2022.10.31 for R2C

walmartshopper, 2022-11-25:
I clean flashed the full build of 21.02 release 2022.10.31 on a NanoPi R2C. I disabled docker and most of the extra daemons, then set up SQM for 22.5Mb up and down.  I ran it for about 24 hours and I had the r8152 driver crash 3 or 4 times where the connection went down for several minutes and then eventually reset.  I'm not running the build anymore so I don't have the exact syslog messages, but it was some error about r8152 and then lan link is down.  I've seen similar stability issues with r8152 when running high throughput, but it was crashing at 22.5Mbps throughput.

hojnikb, 2022-11-26:
I had the same issues with this USB ethernet crashing under heavier loads. Didn't try 22.03 or this custom build, but i've had good luck with github.com/quintus-lab builds It still crashes but rarely. You might want to give it a try.

walmartshopper, 2022-11-26:
Thanks, I went back to my own stock build of 22.03 which has been stable for several months, at least up to 50Mbps, haven't tried any higher.

anaelorlinski, 2022-11-29:
I am not working actively on branch 21.02 anymore because I switched on 22.03 for all my workloads.
However if the same happens in 22.03 let me know. Maybe switching to realtek r8152 would make it more stable

PhatHub, 2023-04-02:
> I am not working actively on branch 21.02 anymore because I switched on 22.03 for all my workloads. However if the same happens in 22.03 let me know. Maybe switching to realtek r8152 would make it more stable

Hi! I attempted to boot with `full-22.03-20230106` on an R2C but unfortunately I can't get it to boot? Admittedly I haven't cracked open the case to connect to the UART but at least for now the LAN side of the network isn't giving me an IP upon DHCP request.

I've been able to use walmartshopper's [image from their Google Drive](https://forum.openwrt.org/t/openwrt-support-for-nanopi-r2c/111620/11) but that image is sorely outdated and none of the opkg updates are compatible anymore. 

If you have any guidance on how I can help give better feedback, I'd love to help out.

anaelorlinski, 2023-04-03:
What I can recommend is to check if one of the previous 22.03 images work, maybe starting this release
https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/releases/tag/OpenWrtAO-22.03-20220428

I don't own a device so I can never check that when I did changes it would still work with R2C. This is also why I will drop support for next branch.

walmartshopper, 2023-04-03:
I have two R2Cs in remote locations.  Will be getting one of them back soon so I might try doing a new build.  The other R2C is running a 22.03-rc3 build which has been stable so far.  I did that build on my laptop which I don't have access to at the moment.  At one point I tried doing a newer build somewhere around 22.03.1 and the driver was crashing so I went back to rc3.  If I can get my R2C back and get a stable build of 22.03.3+ I'll post it on the openwrt forum.

PhatHub, 2023-04-04:
@anaelorlinski Thanks... I'll try them out! 

>  I don't own a device so I can never check that when I did changes it would still work with R2C. This is also why I will drop support for next branch.

I don't blame you. R2C seems to be falling out of favor, and running the patches to enable building R2C openwrt images needs work. 
Also, I can't seem to get more than 90mbps throughput to my 600mbps Internet, regardless of the openwrt install. I was thinking of giving it up to be a wifi/ethernet bridge (for old ethernet-only nodes) but I don't have kmod opkg access to use USB wifi devices... So I might just toss it for an Nanopi R2S.

@walmartshopper 
Hey there... what throughput are you getting on the R2Cs? Were you able to get the Nanopi R2C patch working on the latest openwrt source, or should I just manually add the lines?

anaelorlinski, 2023-06-28:
Support for R2C officially added in 23.05

closed 2023-06-28 by anaelorlinski