anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #41: Can't compile on Ubuntu 22.04

mikegleasonjr, 2023-03-27:
Hi, 

I cleaned installed a new Ubuntu 22.04 build and installed the necessary dependencies and since then I can't compile OpenWRT anymore... Seems like `m4` is causing a bug or something at step 7. All other previous steps are working fine...

```
$ ./openwrt-22.03/steps/07-all_kmods.sh
/OpenWrt-NanoPi-R2S-R4S-Builds
#
# configuration written to .config
#
make[2]: Entering directory '/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/scripts/config'
time: target/linux/prereq#0.08#0.01#0.09
 make[1] target/compile
 make[2] -C target/linux compile
    ERROR: target/linux failed to build.
make -r target/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/include/toplevel.mk:230: target/compile] Error 1
make[2]: Entering directory '/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/scripts/config'
make[1]: Entering directory '/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt'
cd "/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt"; git log --format=%h -1 toolchain > /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/tmp/.ver_check
cmp -s /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/tmp/.ver_check /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.2.0_musl/stamp/.ver_check || { \
	rm -rf /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/staging_dir/target-aarch64_generic_musl /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.2.0_musl /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/toolchain-aarch64_generic_gcc-11.2.0_musl; \
	mkdir -p /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.2.0_musl/stamp; \
	mv /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/tmp/.ver_check /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.2.0_musl/stamp/.ver_check; \
}
make[2]: Entering directory '/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/target/linux'
make[3]: Entering directory '/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/target/linux/rockchip'
rm -f /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/localversion
/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/scripts/kconfig.pl  + /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/target/linux/generic/config-5.10 /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/target/linux/rockchip/armv8/config-5.10 > /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/.config >> /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.target
/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/scripts/package-metadata.pl kconfig /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/tmp/.packageinfo /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/.config 5.10 > /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.override
/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/scripts/kconfig.pl 'm+' '+' /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.target /dev/null /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.override > /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.set
mv /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.set /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.old
grep -v INITRAMFS /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.old > /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.set
rm -rf /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/modules
cmp -s /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.set /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.prev || { cp /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.set /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config; cp /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.set /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.prev; }
export MAKEFLAGS= ; [ -d /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/user_headers ] || make -C /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161 KCFLAGS="-ffile-prefix-map=/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="mike" KBUILD_BUILD_HOST="openwrt" KBUILD_BUILD_TIMESTAMP="Tue Jan  3 00:24:21 2023" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.161 CC="aarch64-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/user_headers headers_install
grep '=[ym]' /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.config.set | LC_ALL=C sort | /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/staging_dir/host/bin/mkhash md5 > /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.vermagic
touch /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/.configured
rm -f /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/vmlinux /OpenWrt-NanoPi-R2S-R4S-Builds/build/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.161/System.map
[... cut ...]
$ which m4
/usr/bin/m4
$ m4 --version                                                                                                                                                                 [main]
m4 (GNU M4) 1.4.18
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Rene' Seindal.
```

mikegleasonjr, 2023-03-27:
I was finally able to compile a version by skipping the step `07` and directly calling `make` in the `build/openwrt` folder...

closed 2023-03-27 by mikegleasonjr

anaelorlinski, 2023-03-29:
Ok thanks for reporting anyway. I know that something wrong might happen with m4 soon