anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #42: Clock speed?

hojnikb, 2023-04-16:
What kind of clock speeds does this firmware offer on R2S/R4S? I'm hesitant to go back to openwrt I'm currently using [quintus-lab] fork, because it has 1.6Ghz clock, which just happens to be enough performance to fully saturate 1Gbit WAN.

anaelorlinski, 2023-04-16:
It's default clock speed, there are no overclocking patches, but optimized compilation flags specific to r2s and r4s CPUs.
You could also check official images for R2S / R4S because support has been streamlined since 22.03 and for your purpose it might be enough.

closed 2023-04-16 by anaelorlinski

hojnikb, 2023-04-16:
> It's default clock speed, there are no overclocking patches, but optimized compilation flags specific to r2s and r4s CPUs. You could also check official images for R2S / R4S because support has been streamlined since 22.03 and for your purpose it might be enough.

cool, i might give stock images another try. With 21.02 i was struggling to hit more than 800-850Mbit WAN speeds, even though my firewall settings were stock and no QoS.

Thanks!