anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #47: Adblocking through AdGuard Home, broken?

wikar, 2023-07-17:
Can't seem to get ad blocking to work since the switch to AdGuard Home. Any instructions on how to get it to work? Have done the initial configuration through port 3000 and followed various guides without any success thus far.

Thanks for the great effort in providing these builds!

anaelorlinski, 2023-07-18:
It works for me, but it does not work out of the box, you need to customize few things in your network and adguardhome config.
I will post a documentation file with my settings, but it's the result of the recommendations of this page : https://openwrt.org/docs/guide-user/services/dns/adguard-home

Mincka, 2023-08-26:
Can you share you settings @anaelorlinski ? Thank you. :)

anaelorlinski, 2023-08-26:
Here is a quick summary of MY settings, they differ a bit from the setup done on https://openwrt.org/docs/guide-user/services/dns/adguard-home so keep it in mind :)
Let me know if you get success, I will add this documentation to the repo

I am using Quad9 DNS. So the IPs correspond to the ones you can find here : https://adguard-dns.io/kb/general/dns-providers/#quad9-dns

Update dnsmasq port to be 5353 and not be main DNS server anymore, and deliver router IP as the DNS resolver to DHCP clients :
In **/etc/config/dhcp**
```
config dnsmasq
        ... other settings
        # setting for adguardhome
	option rebind_protection '0'
	option port '5353'

config dhcp 'lan'
	... other settings
	list dhcp_option '3,192.168.1.1'   #3,routerIP
	list dhcp_option '6,192.168.1.1'   #6,routerIP
```

Configure AGH to make it the main DNS server. Here are the relevant settings in adguardhome.yaml, you might be able to update some of them through the GUI but from what I remember I had to setup things manually here.

In **/etc/adguardhome.yaml** (only relevant settings are shown)
```
bind_host: 192.168.1.1
bind_port: 8080
clients:
  runtime_sources:
    rdns: true       #GUI : Enable Reverse resolving of client's IP adresses
dns:
  bind_hosts:
    - 192.168.1.224
    - 127.0.0.1
  port: 53
  upstream_dns:                            #GUI : upstream DNS servers
    - '[/lan/]127.0.0.1:5353'
    - '[//]127.0.0.1:5353'
    - '[/pool.ntp.org/]1.1.1.1'
    - '[/pool.ntp.org/]1.0.0.1'
    - '[/pool.ntp.org/]2606:4700:4700::1111'
    - '[/pool.ntp.org/]2606:4700:4700::1001'
    - https://dns.quad9.net/dns-query
  bootstrap_dns:                         # GUI : bootstrap DNS servers
    - 9.9.9.10
    - 149.112.112.10
    - 2620:fe::10
    - 2620:fe::fe:10
  use_private_ptr_resolvers: true    #GUI : Use private DNS resolvers
  local_ptr_upstreams:
    - 127.0.0.1:5353                          #GUI : private reverse DNS servers
  ```

Notes :
- Router dns resolution is not done through AGH
- IPv6 DNS adresses are not dispatched to DHCP clients

Mincka, 2023-08-26:
Thanks a lot. I've just forked your repo and I try to build an image with a larger `CONFIG_TARGET_ROOTFS_PARTSIZE` because I don't want to resize it after the installation.
I'll do the setup of `AdGuard` once everything will be running smoothly. Thank you for this repo!

anaelorlinski, 2023-08-26:
Seems a good idea to make a bigger rootfs, since all the SD cards are now minimum 32Gb.
I will add on my TODO list

closed 2024-03-27 by anaelorlinski