anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #48: Possible to update LuaLanes or add patch in support of sqm-autorate?

lwndow, 2023-08-12:
Hello,

Thanks for your work in supporting the NanoPi's! I've came across [sqm-autorate](https://github.com/sqm-autorate/sqm-autorate) which I'd like to use, but it seems there's [an issue](https://github.com/sqm-autorate/sqm-autorate/issues/32) with pthreads and LuaLanes in the current package available to OpenWRT.

There's [a patch](https://github.com/Fail-Safe/sqm-autorate/blob/testing/lua-threads/patch/300-fix-gnu-c-pthread_yield-error.patch) listed that I'm wondering if can be evaluated for inclusion in the AO builds to let us have a functioning autorate? Or perhaps an update of LuaLanes as [described here](https://github.com/sqm-autorate/sqm-autorate/issues/32#issuecomment-1272164485)? 

Be great for those of us with variable pools (read: Cable) connections!

Thanks for considering!

anaelorlinski, 2023-08-13:
Sorry but after checking it's not in my plans to add this package or patch to my builds.
Probably try https://github.com/lynxthecat/cake-autorate instead that seems to achieve same goal without lua.
Else I can suggest you to rebuild the failing package with a OpenWrt SDK

closed 2023-08-13 by anaelorlinski

Fail-Safe, 2024-01-21:
@lwndow I am hoping to have this rectified soon per https://github.com/openwrt/packages/pull/23121.