anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #49: Support for build failure

Mincka, 2023-08-26:
Hello,
I've forked your repo and tried to build R4S full seed on 22.03 branch.
I tried against main and the tagged version for 22.03.5.

The only update I did was to set `CONFIG_TARGET_ROOTFS_PARTSIZE=31744` to extend to rootfs partion on a Micro SD Card of 32 GB.

I don't understanrd what could cause the difference.
It seems related to `include/config/auto.conf.cmd`

```
SYNC    include/config/auto.conf.cmd
Error in reading or end of file.
```

My build where it fails
https://github.com/Mincka/OpenWrt-NanoPi-R2S-R4S-Builds/actions/runs/5984628256/job/16235981702#step:18:79

Yours at the same step
https://github.com/anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds/actions/runs/5942047886/job/16114172364#step:23:79

Thanks!

anaelorlinski, 2023-08-26:
From the logs I would say it's linked to 
Promise SATA SX8 support (BLK_DEV_SX8) [N/m/y/?] (NEW) missing somewhere in config.
Might need to be fixed similar to this : https://github.com/anaelorlinski/OpenWrt-NanoPi-R5S-Builds/blob/e3ec1f47fdae220d29310f4792ceb8dabcca06a2/openwrt-23.05/steps/03_patch_openwrt.sh#L65
I will try on my side

anaelorlinski, 2023-08-27:
Quick status, I don't get the same error but get disk space issues, because the 32gb images need to be written to disk.
So sounds not so easy to build with GH actions, because the disk space is already hitting the limit with regular builds.

Mincka, 2023-08-28:
Oh yes, indeed, when the image will be uncompressed, it'll take 32 GB since it's the full image to flash.
It does not seem to be a good idea... I guess I'll stick with post installation steps and just drop this idea.

It's still strange that I don't get the same results as you by forking your project and workflows.
I tried to add a seed for the specific option you've highlighted by [mimicking what was done for other settings](https://github.com/Mincka/OpenWrt-NanoPi-R2S-R4S-Builds/commit/7190921b2f1e262966da66972a03741adc136ff3), but no luck, same result.

I think your approach to prepare an up-to-date SD Card with post actions using GitHub and swap it on the Pi for upgrades is the best option. It's just a bit "sad" to waste 30 GB of memory but sticking with squashfs on SD Card, and use external storage for Docker containers for instance, should be more resilient over the time.

Do you have this GitHub workflow publicly available somewhere? I don't remember where I saw you speaking about that, either in another issue, or on the RS4 thread ofthe OpenWRT forum.

closed 2024-03-27 by anaelorlinski