anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #50: 22.03.5 full on R4S - can't install wireless USB card driver

blackie333, 2023-09-19:
Hello, 
installation of your latest 22.03.5 full version for R4S and few extra packages (apcupsd, unbound, banip) worked fine.
Today I received my new USB wifi adaptor (mt7612u based ALFA AWUS036ACM) so I tried to install missing module kmod-mt76-usb via Luci, but received following error:
The installed version of package kernel is not compatible, require 5.10.176-1-9ba0cc12… while 5.10.176-1-39f97d6e… is installed.
opkg also refused to install kmod-mt76-usb because of kernel version.

``` opkg install kmod-mt76-usb
Unknown package 'kmod-mt76-usb'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.176-1-9ba0cc12a8e6bbf68463d91f0091bf15) for kmod-cfg80211
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-cfg80211 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.176-1-9ba0cc12a8e6bbf68463d91f0091bf15) for kmod-mac80211
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-mac80211 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.176-1-9ba0cc12a8e6bbf68463d91f0091bf15) for kmod-mt76-core
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-mt76-core found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.176-1-9ba0cc12a8e6bbf68463d91f0091bf15) for kmod-mt76-usb
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-mt76-usb found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-mt76-usb.
```
Was under impression your build is based on 22.03.5 stable kernel so I haven't expected complications.
What can I do to install the required modules? I don't know how to compile own modules :(
Thank you in advance for help.

anaelorlinski, 2023-09-19:
Hello!
The complications always come from kernel modules because they are not compatible with a different build (it works for regular packages). I partially worked around the problem by including the default built kernel modules in the /ipks folder but the ones built inside packages are not included. Unfortunately this is the case for the wifi kmods you need.
I have no immediate fix for this. I will check if this is feasible to add them somehow.
Anyway my builds for R4S are no longer very relevant compared to vanilla openwrt since patches to support the device are now integrated. I would suggest to give a try to the vanilla openwrt and you should be able to add the packages without issues.

blackie333, 2023-09-20:
OK, thank you for trying to help me. 
I have chosen your build because it was advertised in the R4S thread as the best containing speed tweaks and most compatible realtek ethernet driver. 
So you suggest to use vanilla 22.03.5 stable build - but I'm not sure whether it can reach the same routing speed?
That is not bad idea but sadly I have invested 2 days of configuration into your build :(
It would be easier if the driver could be installed into your build somehow...

joopdo, 2024-04-18:
Came here with the same issue. Falling back to vanilla 23.05.03 now.

anaelorlinski, 2024-04-18:
Yes I think my builds are not relevant anymore since the support in vanilla openwrt seems now pretty solid, and kernel modules are not compatible between different builds (for good reasons).