anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #51: Please Update To Latest Build

adibindrianto, 2024-03-26:
Need update to openwrt 23.05.3 latest update, your repo is amazing for me

adibindrianto, 2024-03-27:
[image]
i just changed the rules of some packages in ao-R4S-full.seed.orig and ao-R4S-full.seed but i want to build for R2S because in the repo there are no rules for changing R2S packages so i changed in R4S, but it always fails like picture, can you help me?

anaelorlinski, 2024-03-27:
The kernel config needs some refresh too because the build does not have info about NET_SCH_CBQ
You need to add the line
```
# CONFIG_NET_SCH_CBQ is not set
```
in openwrt-23.05/steps/03_patch_openwrt.sh at line 55

anaelorlinski, 2024-03-27:
build has completed, I will try it on my router tonight

closed 2024-03-27 by anaelorlinski

adibindrianto, 2024-03-27:
I also tried it and succeeded, thank you

anaelorlinski, 2024-03-28:
please read openwrt documentation first or ask in the openwrt forum.
you would find easily that you just need to add this line to /etc/opkg/customfeeds.conf

adibindrianto, 2024-03-28:
[image]

understand, I mean is there an additional script line for the custom feed before the build is run, maybe there is a script that needs to be added to the bottom line for the custom feed, I understand that this is vanilla openwrt if you want to add a custom feed line can you add the bottom line? and how should the script be executed?

anaelorlinski, 2024-03-28:
The url you show contains .ipk files, already compiled packages.
So flash the cart and afterwards add "src/gz openwrt_kiddin9 https://dl.openwrt.ai/latest/packages/aarch64_generic/kiddin9" to /etc/opkg/customfeeds.conf
then run opkg update
and opkg install your extra packages.
If you want to add source packages to the build and build them, the way to do it is totally different, check how I integrate the stangri repository from source into my build.

adibindrianto, 2024-03-28:
I'm sorry, I have followed as far as I can but when the script makes me confused then that's where I have to stop because there is a lot I don't know about the build script but thank you for responding to all my complaints

adibindrianto, 2024-03-30:
I want to report an issue, after the first time I finished burning, Docker didn't work and couldn't display the web UI, for example Jellyfin

adibindrianto, 2024-03-30:
Do you have a desire to make immortalwrt the same as openwrt r2s? with the latest version of vanilla immortalwrt?