anaelorlinski/OpenWrt-NanoPi-R2S-R4S-Builds #52: R2S Plus emmc install

dgensem, 2024-03-27:
Hi,
 Would you be able to give the coordinates on how to install your vanilla openwrt image in the internal r2s plus emmc memory?

anaelorlinski, 2024-03-27:
Some users reported success with those instructions : 
https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R5S#Option_2:_Install_OS_on_Web_Page

closed 2024-03-27 by anaelorlinski

dgensem, 2024-03-27:
Tried ext4 and squashfs, no go using emmc flash tool.

this is the error I get in the serial console.

U-Boot SPL 2022.07-OpenWrt-r23630-842932a63d (Nov 14 2023 - 13:38:11 +0000)
Trying to boot from MMC1
Card did not respond to voltage select! : -110
spl: mmc init failed with error: -95
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###

anaelorlinski, 2024-03-27:
Please try using vanilla images from there : https://downloads.openwrt.org/releases/23.05.3/targets/rockchip/armv8/
Not sure that my UBoot supports booting from eMMC, and I will not update it because it would break booting on 1GB R4S